kassane/zig-esp-idf-sample #1: Roadmap and FAQs

kassane, 2024-03-19:
## TODO

- [x] bluetooth
- [x] lwip
- [ ] peripherals (led, gpio, adc, uart, etc...)
- [x] netif
- [x] mqtt 
- [x] wifi
- [ ] tls (mbedtls)
- [x] freertos (tasks, queue,semaphores, etc...) 

## mini-FAQ

- **So Zig can read C headers. Why manual wrappers?**
Indeed `@cImport`/`@cInclude` doesn't access C types directly, but it converts to zig binding automatically (like binding-rs).
However, there have been some limits to support at the moment, in particular when converting macros (difficult to parse).
One example is `ESP_LOG`x, which has been done manually, since the original version provided in `esp_log.h` has become untranslatable.
    **see:** [zig doc - `@cImport` vs `translate-c`](https://ziglang.org/documentation/master/#cImport-vs-translate-c)

- **So why not use [microzig](https://microzig.tech/) to make the project fully zig-written?**
I am currently using zig v0.12.0-dev (nightly/master) and the **microzig** project does not support this version yet.
    **see:** https://github.com/ZigEmbeddedGroup/microzig/pull/178
The use of regz (svd converter) will also need to be evaluated, since it is under development and not all peripherals are guaranteed during translation.
   **see:** https://github.com/ZigEmbeddedGroup/microzig/pull/180

kassane, 2024-03-20:
Although the readme claims to support v4.4 to 5.x, it is actually being tested on 5.3-dev.

The last commit added a wrapper to handle the IDF version. The main idea was to read the [esp_idf_version.h header](https://github.com/espressif/esp-idf/blob/master/components/esp_common/include/esp_idf_version.h), but zig couldn't find it even though it had access to the include path.

https://github.com/kassane/zig-esp-idf-sample/blob/main/imports%2Fidf-sys.zig#L6-L48

The idea of managing multiple versions of functions could be done in a similar way to crc_xx, depending on the version of the IDF.

https://github.com/kassane/zig-esp-idf-sample/blob/main/imports%2Fidf-sys.zig#L335-L342

### Reference
- https://github.com/esp-rs/esp-idf-template/issues/188

kassane, 2024-04-20:
New tarballs: https://github.com/kassane/zig-espressif-bootstrap/releases/tag/0.12.0-xtensa added
commit: https://github.com/kassane/zig-esp-idf-sample/commit/abcef439b5672b585de82955c8c6a5850d731230

dattodroid, 2024-04-26:
Hello, how to build examples/mqtt_client.zig ?  I'm getting the following error:
`~/zig/zig-esp-idf-sample/examples/mqtt_client.zig:4:46: error: root struct of file 'idf' has no member named 'esp_mqtt_event_handle_t'`

(I did not investigate at all)