kassane/zig-esp-idf-sample #13: Zig tracking issues

kassane, 2024-08-20:
- [ ] https://github.com/ziglang/zig/issues/20663
- [ ] https://github.com/ziglang/zig/issues/20630
- [ ] https://github.com/ziglang/zig/issues/20327

TheDarkUndoing, 2026-10-06:
Is this tracker still active, now that the zig code base is on codeberg?

kassane, 2026-10-06:
The Zig dev team has started using Github for legacy issue handling.
Mentioned here: https://ziglang.org/download/0.17.0/release-notes.html#Bug-Fixes

Also, llvm upstream merge progress..

> -  https://github.com/espressif/llvm-project/issues/4

### Codeberg:

> - [Allow overriding `std.Io.File`, `std.Io.Dir`](https://codeberg.org/ziglang/zig/issues/30604)
> - [std: os independency day]https://codeberg.org/ziglang/zig/pulls/31235)

Any interesting proposals that influence the project are relevant and should be reported.