kassane/zig-esp-idf-sample #16: Build in DevContainer fails

JanBeelte, 2024-11-12:
Hi all,

I am trying to get my ESP32-C3 programmed with zig. I wanted to start out from the DevContainer to minimize issues with my system setup. If I call:
`zig build -Dtarget=riscv32-freestanding-none -Dcpu=generic_rv32+c+m+zicsr+zifencei` from within the DevContainer this is what I get:
```
esp@2c44f4cee9f4:/workspaces/open-sensor-platform/firmware/blink$ zig build -Dtarget=riscv32-freestanding-none -Dcpu=generic_rv32+c+m+zicsr+zifencei
error: FileNotFound
/opt/zig-relsafe-espressif-x86_64-linux-musl-baseline/lib/std/posix.zig:1791:23: 0x148bd4d in openatZ (build)
            .NOENT => return error.FileNotFound,
                      ^
/opt/zig-relsafe-espressif-x86_64-linux-musl-baseline/lib/std/fs/Dir.zig:1567:21: 0x144afe6 in openDirFlagsZ (build)
        else => |e| return e,
                    ^
/opt/zig-relsafe-espressif-x86_64-linux-musl-baseline/lib/std/fs/Dir.zig:1531:5: 0x13fe502 in openDirZ (build)
    return self.openDirFlagsZ(sub_path_c, symlink_flags);
    ^
/opt/zig-relsafe-espressif-x86_64-linux-musl-baseline/lib/std/fs/Dir.zig:1475:5: 0x13f290e in openDir (build)
    return self.openDirZ(&sub_path_c, args);
    ^
/workspaces/open-sensor-platform/firmware/blink/build.zig:86:15: 0x1412a98 in searched_idf_libs (build)
    var dir = try std.fs.cwd().openDir("../build", .{
              ^
/workspaces/open-sensor-platform/firmware/blink/build.zig:35:9: 0x14132f3 in includeDeps (build)
        try searched_idf_libs(b, lib);
        ^
/workspaces/open-sensor-platform/firmware/blink/build.zig:19:5: 0x14141e9 in build (build)
    try includeDeps(b, lib);
    ^
/opt/zig-relsafe-espressif-x86_64-linux-musl-baseline/lib/std/Build.zig:2168:25: 0x13f6607 in runBuild__anon_4612 (build)
        .error_union => try build_zig.build(b),
                        ^
/opt/zig-relsafe-espressif-x86_64-linux-musl-baseline/lib/compiler/build_runner.zig:335:9: 0x13f135b in main (build)
        try builder.runBuild(root);
        ^
error: the following build command failed with exit code 1:
/workspaces/open-sensor-platform/firmware/blink/.zig-cache/o/2aed623c67fa95b300a7d5bd2356f4c0/build /opt/zig-relsafe-espressif-x86_64-linux-musl-baseline/zig /opt/zig-relsafe-espressif-x86_64-linux-musl-baseline/lib /workspaces/open-sensor-platform/firmware/blink /workspaces/open-sensor-platform/firmware/blink/.zig-cache /home/esp/.cache/zig --seed 0x15fba01 -Zbcbe1850831e84be -Dtarget=riscv32-freestanding-none -Dcpu=generic_rv32+c+m+zicsr+zifencei
```

What am I doing wrong?

Thanks for your help :-)

kassane, 2024-11-12:
Hi @JanBeelte,

The error says file not found in your project.
You may need to check `build.zig` if it is missing user file.

Do you have any open project that can replicate this problem?

JanBeelte, 2024-11-12:
Thank you so much for the quick response!

> Do you have any open project that can replicate this problem?

https://github.com/opendata-nuernberg/open-sensor-platform/tree/main/firmware/blink

We are trying to use your sample as a starting point for our firmware. If there is any better way to use your "toolchain" as a dependency (instead of copying it into our repo) advice is very welcome!

JanBeelte, 2024-11-12:
Update, I just discovered `idf.py build`
Is this the correct way to build the project? Why is it not mentioned in the readme?
This is the output I get:
```
esp@2c44f4cee9f4:/workspaces/open-sensor-platform/firmware/blink$ idf.py build
Executing action: all (aliases: build)
Running ninja in directory /workspaces/open-sensor-platform/firmware/blink/build
Executing "ninja all"...
[1/1] cd /workspaces/open-sensor-platform/firmware/blink/bu...sor-platform/firmware/blink/build/bootloader/bootloader.bin
Bootloader binary size 0x6590 bytes. 0xa70 bytes (9%) free.
[5/7] Linking CXX executable zig-sample-idf.elf
/opt/esp/tools/xtensa-esp-elf/esp-14.2.0_20240906/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld: warning: /opt/esp/tools/xtensa-esp-elf/esp-14.2.0_20240906/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/esp32/no-rtti/crtn.o: missing .note.GNU-stack section implies executable stack
/opt/esp/tools/xtensa-esp-elf/esp-14.2.0_20240906/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/14.2.0/../../../../xtensa-esp-elf/bin/ld: NOTE: This behaviour is deprecated and will be removed in a future version of the linker
[6/7] Generating binary image from built executable
esptool.py v4.8.1
Creating esp32 image...
Merged 2 ELF sections
Successfully created esp32 image.
Generated /workspaces/open-sensor-platform/firmware/blink/build/zig-sample-idf.bin
[7/7] cd /workspaces/open-sensor-platform/firmware/blink/bu...pen-sensor-platform/firmware/blink/build/zig-sample-idf.bin
zig-sample-idf.bin binary size 0x36210 bytes. Smallest app partition is 0x100000 bytes. 0xc9df0 bytes (79%) free.

Project build complete. To flash, run:
 idf.py flash
or
 idf.py -p PORT flash
or
 python -m esptool --chip esp32 -b 460800 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_size 2MB --flash_freq 40m 0x1000 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0x10000 build/zig-sample-idf.bin
or from the "/workspaces/open-sensor-platform/firmware/blink/build" directory
 python -m esptool --chip esp32 -b 460800 --before default_reset --after hard_reset write_flash "@flash_args"
```

zig build is giving the same error as before.

kassane, 2024-11-12:
> (instead of copying it into our repo) advice is very welcome!

Yes!😅 I haven't made the project an idf-component yet. PRs are welcome!!

Maybe help you using `zon`:
https://github.com/georgik/esp32-lang-lab/tree/main/examples/zig/esp-idf-v5/get-started/hello_world

However, need add CMake script.

---

Regarding the highlighted error, I need to test it. Unfortunately, I can't at this exact moment!!

---

Answer to another question about "my toolchain fork"...

First of all, feel free to use it.

However, the credit goes to Zig upstream. I just anticipated Xtensa's tests!

Recommended if `xtensa` target is needed, `riscv32` usage may be optional (zig upstream already support).

kassane, 2024-11-12:
> > The error says file not found in your project. You may need to check `build.zig` if it is missing user file.
> 
> I checked the build.zig file:
> [image]
> looks like it is searching for another build.zig file outside of the repository, or do I miss something here?

It's not another file it's CMake's view between src_dir and build_dir

kassane, 2024-11-12:
See
https://github.com/kassane/zig-esp-idf-sample/blob/main/cmake%2Fzig-build.cmake#L89-L104

JanBeelte, 2024-11-12:
Thank you so much for all the input!
The current state is:
`idf.py build`
is generating a succesful build.
But still `zig build` is throwing the same error that we started with.
I did not find any mention of `idf.py` in your Readme, is this an "internal" tool or is it supposed to be used by "users"?
I would really appreciate if you could take a look why our zig build fails whenever you have time :-)

kassane, 2024-11-12:
> I did not find any mention of `idf.py` in your Readme, is this an "internal" tool or is it supposed to be used by "users"?

I assume you are esp-idf user.

`build.zig` is not the "protagonist" in the build, continuing the use of CMake.
CMake calls `zig build` during the build process.

Do not run Zig standalone. Except for zls/zigscient with `idf` EnvVar enabled to get imports.

Zig becomes secondary in library building by integrating `app_main` into the source.

JanBeelte, 2024-11-12:
This explains it very well, thank you very much :-)