kassane/zig-esp-idf-sample #19: [Feature Request] Refactoring into idf library and separate blink example

JanBeelte, 2024-11-27:
We are thinking about refactoring the codebase in the following directory structure:
1. idf/
 - imports
 - includes
 - ...
2. blink/
 - app.zig
 - build.zig
 - build.zon

What do you think about this structure @kassane ?
My hope is if we use this structure we could reference the idf-library directly from our build.zon midterm so we would not need to include it in our codebase directly.
I am envisioning sth like this:
```
    .dependencies = .{
        .idf = .{
            .url = "git+https://github.com/kassane/zig-esp-idf-sample/idf",
            .hash = "",
        },
    },
```

If you would agree that this structure is a good way forward I am happy to help with the refactoring (I would need your guidance with respect to the idf / cmake integration though)

kassane, 2024-11-27:
The first time it was that the api shared was this:

https://github.com/georgik/esp32-lang-lab/tree/main/examples%2Fzig%2Fesp-idf-v5%2Fget-started%2Fhello_world

---

However, I still need to be careful with cmakelists. How would this refactoring work?

Unfortunately the current solution for this project depends on CMake!!

If the proposal makes it easier and passes the CI/CD test, I approve it without any problem.

kassane, 2024-11-28:
> I would need your guidance with respect to the idf/cmake integration though

Zig's integration into cmake is modularized (see https://github.com/kassane/zig-esp-idf-sample/commit/a313db9c13a9bb827791041c56cdb547cc6ed2c2) and detached from the `main/CMakeLists.txt` script, perhaps making it easier to include in multiple cmakelists (`add_subdirectory(blink_path)` projects).

JanBeelte, 2024-11-30:
Hey @kassane,
I did some initial tests of separating the two projects. My idea would be to only rely on zig build system for separation, make zig build system pull in the dependency and then provide the final build result to cmake.
Unfortunately I have some issues with the setup that I do not fully understand.
Here is my initial setup:
zig-idf base library:
https://github.com/opendata-nuernberg/zig-idf
zig-idf application sample:
https://github.com/JanBeelte/zig-esp-idf-sample

if you check out both repositories on the same directory level e.g.:
~/projects/zig-esp-idf-sample
 ~/projects/zig-idf
you should be able to run `idf.py build` from within zig-esp-idf-sample. The dependency resolution already works and the zig-idf project is compiled. During that process you can see a lot of errors from zig build system complaining that it cannot find the imports so it looks like I broke sth with respect to the build system I currently do not fully understand.
Maybe you could take a look and provide a hint?

JanBeelte, 2024-11-30:
Update, I managed to fix all the import issues, now I am facing a linking issue:
```
libapp_zig.a.o):(.literal+0xc0): undefined reference to `esp_log_system_timestamp'
collect2: error: ld returned 1 exit status
```
do you maybe have an idea how to fix that?

kassane, 2024-12-01:
> anything that I should change here?

Remove or replace `esp_log_system_timestamp` -- compare to `import/log.zig` in esp_log function and see `esp_log_timestamp`.

JanBeelte, 2024-12-02:
Ok we have a working proposal for the restructuring now.
We split the repository into two parts, a base library and some minimal samples that show how to use it.

You can find our proposal for the base library here:
https://github.com/opendata-nuernberg/zig-idf
If you want we can keep the ownership in opendata-nuernberg and invite you as an admin or you create a new repository where you have exclusive ownership.
Whatever you prefer, I would propose to remove the word "sample" from the repository name.

You can find our proposal for the samples here:
https://github.com/JanBeelte/zig-esp-idf-sample
I also added some instructions on how to build the blink example for beginners.
What I really like about this new approach that we can fully utilise the zig build system for the base library like this:
https://github.com/JanBeelte/zig-esp-idf-sample/blob/fb49237674eab08be5249aeb7fedd8c4fe178493/blink/build.zig.zon#L12

I also tried to create working versions of blink_s3SPI and mqtt_client that I found in your example folder. Unfortunately I could not make them compile, maybe you can chime in here?

kassane, 2024-12-02:
> If you want we can keep the ownership in opendata-nuernberg and invite you as an admin or you create a new repository where you have exclusive ownership.

No, thank you! However, PRs are accepted from this proposal up.

> I also tried to create working versions of blink_s3SPI and mqtt_client that I found in your example folder. Unfortunately I could not make them compile, maybe you can chime in here?

`Mqtt` is incomplete [untested], I admit (based on the official example)!!!
`esp32s3_blink` was a RGB_led test. I tried tested it both in C++ and Zig with `build.zig` and my device worked, although it is still subject to problems from upstream (esp-idf). See #3

---

Besides those points, you're doing a great refactoring process.
Although this project was a solo experiment to sharpen my interest in it (inspired by the esp-rs team).

closed 2025-01-17 by kassane