kassane/zig-esp-idf-sample #22: C6 Wifi Compile Errors

joseph-montanez, 2025-02-02:
I had to make some changes for station (wifi acting as client) and having issues with missing symbols

```bash
: && /Users/josephmontanez_1/.espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/riscv32-esp-elf-g++ -march=rv32imac_zicsr_zifencei -nostartfiles -march=rv32imac_zicsr_zifencei --specs=nosys.specs    -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32C6=0 -Wl,--Map=/Users/josephmontanez_1/Documents/zig-esp-idf-sample/build/zig-sample-idf.map -Wl,--no-warn-rwx-segments -Wl,--orphan-handling=warn -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T rom.api.ld -T esp32c6.peripherals.ld -T esp32c6.rom.ld -T esp32c6.rom.api.ld -T esp32c6.rom.rvfp.ld -T esp32c6.rom.wdt.ld -T esp32c6.rom.systimer.ld -T esp32c6.rom.version.ld -T esp32c6.rom.phy.ld -T esp32c6.rom.coexist.ld -T esp32c6.rom.net80211.ld -T esp32c6.rom.pp.ld -T esp32c6.rom.newlib.ld -T esp32c6.rom.newlib-normal.ld -T esp32c6.rom.heap.ld -T memory.ld -T sections.ld CMakeFiles/zig-sample-idf.elf.dir/project_elf_src_esp32c6.c.obj -o zig-sample-idf.elf -L/Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/riscv/ld   -L/Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/soc/esp32c6/ld   -L/Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/esp_rom/esp32c6/ld   -L/Users/josephmontanez_1/Documents/zig-esp-idf-sample/build/esp-idf/esp_system/ld   -L/Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/esp_phy/lib/esp32c6   -L/Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/esp_coex/lib/esp32c6   -L/Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/esp_wifi/lib/esp32c6 esp-idf/riscv/libriscv.a  esp-idf/esp_driver_gpio/libesp_driver_gpio.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/esp_app_format/libesp_app_format.a  esp-idf/esp_bootloader_format/libesp_bootloader_format.a  esp-idf/app_update/libapp_update.a  esp-idf/esp_partition/libesp_partition.a  esp-idf/efuse/libefuse.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_mm/libesp_mm.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_security/libesp_security.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/pthread/libpthread.a  esp-idf/cxx/libcxx.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/esp_driver_gptimer/libesp_driver_gptimer.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/esp_driver_uart/libesp_driver_uart.a  esp-idf/app_trace/libapp_trace.a  esp-idf/esp_event/libesp_event.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/esp_driver_pcnt/libesp_driver_pcnt.a  esp-idf/esp_driver_spi/libesp_driver_spi.a  esp-idf/esp_driver_mcpwm/libesp_driver_mcpwm.a  esp-idf/esp_driver_i2s/libesp_driver_i2s.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/esp_driver_sdspi/libesp_driver_sdspi.a  esp-idf/esp_driver_sdio/libesp_driver_sdio.a  esp-idf/esp_driver_rmt/libesp_driver_rmt.a  esp-idf/esp_driver_tsens/libesp_driver_tsens.a  esp-idf/esp_driver_sdm/libesp_driver_sdm.a  esp-idf/esp_driver_i2c/libesp_driver_i2c.a  esp-idf/esp_driver_ledc/libesp_driver_ledc.a  esp-idf/esp_driver_parlio/libesp_driver_parlio.a  esp-idf/esp_driver_usb_serial_jtag/libesp_driver_usb_serial_jtag.a  esp-idf/driver/libdriver.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/esp_vfs_console/libesp_vfs_console.a  esp-idf/vfs/libvfs.a  esp-idf/lwip/liblwip.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_coex/libesp_coex.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/unity/libunity.a  esp-idf/cmock/libcmock.a  esp-idf/console/libconsole.a  esp-idf/http_parser/libhttp_parser.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc/libesp_adc.a  esp-idf/esp_driver_cam/libesp_driver_cam.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/esp_https_server/libesp_https_server.a  esp-idf/esp_lcd/libesp_lcd.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/protocomm/libprotocomm.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/fatfs/libfatfs.a  esp-idf/ieee802154/libieee802154.a  esp-idf/json/libjson.a  esp-idf/mqtt/libmqtt.a  esp-idf/nvs_sec_provider/libnvs_sec_provider.a  esp-idf/rt/librt.a  esp-idf/spiffs/libspiffs.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/main/libmain.a  esp-idf/app_trace/libapp_trace.a  esp-idf/unity/libunity.a  esp-idf/esp_https_server/libesp_https_server.a  -u esp_system_include_coredump_init  esp-idf/wear_levelling/libwear_levelling.a  -u nvs_sec_provider_include_impl  esp-idf/protocomm/libprotocomm.a  esp-idf/console/libconsole.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/json/libjson.a  esp-idf/riscv/libriscv.a  esp-idf/esp_driver_gpio/libesp_driver_gpio.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/esp_app_format/libesp_app_format.a  esp-idf/esp_bootloader_format/libesp_bootloader_format.a  esp-idf/app_update/libapp_update.a  esp-idf/esp_partition/libesp_partition.a  esp-idf/efuse/libefuse.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_mm/libesp_mm.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_security/libesp_security.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/pthread/libpthread.a  esp-idf/cxx/libcxx.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/esp_driver_gptimer/libesp_driver_gptimer.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/esp_driver_uart/libesp_driver_uart.a  esp-idf/esp_event/libesp_event.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/esp_driver_pcnt/libesp_driver_pcnt.a  esp-idf/esp_driver_spi/libesp_driver_spi.a  esp-idf/esp_driver_mcpwm/libesp_driver_mcpwm.a  esp-idf/esp_driver_i2s/libesp_driver_i2s.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/esp_driver_sdspi/libesp_driver_sdspi.a  esp-idf/esp_driver_sdio/libesp_driver_sdio.a  esp-idf/esp_driver_rmt/libesp_driver_rmt.a  esp-idf/esp_driver_tsens/libesp_driver_tsens.a  esp-idf/esp_driver_sdm/libesp_driver_sdm.a  esp-idf/esp_driver_i2c/libesp_driver_i2c.a  esp-idf/esp_driver_ledc/libesp_driver_ledc.a  esp-idf/esp_driver_parlio/libesp_driver_parlio.a  esp-idf/esp_driver_usb_serial_jtag/libesp_driver_usb_serial_jtag.a  esp-idf/driver/libdriver.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/esp_vfs_console/libesp_vfs_console.a  esp-idf/vfs/libvfs.a  esp-idf/lwip/liblwip.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_coex/libesp_coex.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/http_parser/libhttp_parser.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc/libesp_adc.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/mbedtls/mbedtls/library/libmbedtls.a  esp-idf/mbedtls/mbedtls/library/libmbedcrypto.a  esp-idf/mbedtls/mbedtls/library/libmbedx509.a  esp-idf/mbedtls/mbedtls/3rdparty/everest/libeverest.a  esp-idf/mbedtls/mbedtls/3rdparty/p256-m/libp256m.a  /Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/esp_coex/lib/esp32c6/libcoexist.a  /Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/esp_wifi/lib/esp32c6/libcore.a  /Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/esp_wifi/lib/esp32c6/libespnow.a  /Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/esp_wifi/lib/esp32c6/libmesh.a  /Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/esp_wifi/lib/esp32c6/libnet80211.a  /Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/esp_wifi/lib/esp32c6/libpp.a  /Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/esp_wifi/lib/esp32c6/libsmartconfig.a  /Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/esp_wifi/lib/esp32c6/libwapi.a  esp-idf/riscv/libriscv.a  esp-idf/esp_driver_gpio/libesp_driver_gpio.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/esp_ [... cut ...]
    idf.wifi.start() catch {};
    idf.wifi.connect() catch {};
```

Note **idf.sys** does not exist I had to add this to the module and include into the build command which is just "idf-sys.zig". Also I know the code above is not correct but I wanted to get the compiling errors fixed first. I tried `idf.py menuconfig` but everything around wifi and WPA was already checked. I ran a C version and it worked find with the same `menuconfig`. Nothing in the CMakefile was needed to be added like I tried.

kassane, 2025-02-02:
Hi @joseph-montanez,

What's IDF version? (I assume it's 5.4)

Can you confirm if this happens always or first time?

The **build.zig** in `searched_idf_libs` will try to capture build object files and not static-libraries from **esp-idf**. This is to increase the size of `libapp_zig`.
The final binary is still the responsibility of **CMake**. It will pull the `libapp_zig` [main function included] before building the binary.
https://github.com/kassane/zig-esp-idf-sample/blob/0085d8160adca587af512bb5d94955d9c79c2c27/build.zig#L85

However, zig will not always run in the order of the **CMake** dependencies (if you have ninja-build installed, it will parallelize the build and change the order of the dependencies that are generated or downloaded).

I suggest trying the rebuild directly (without cleaning) if the first attempt fails.
But if the error continues, the problem is that the required dependencies are not available in object file.

> Note **idf.sys** does not exist I had to add this to the module and include into the build command which is just "idf-sys.zig".

Yeah! Not imported to `esp_idf` pub module, but available in priv dep modules.
https://github.com/kassane/zig-esp-idf-sample/blob/0085d8160adca587af512bb5d94955d9c79c2c27/build.zig#L474-L480

joseph-montanez, 2025-02-03:
Yes it's IDF 5.4. If I build a second time, it still an issue but not inside Zig:

/Users/josephmontanez_1/.espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/../lib/gcc/riscv32-esp-elf/14.2.0/../../../../riscv32-esp-elf/bin/ld: /Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/wpa_supplicant/src/rsn_supp/wpa.c:2373:(.text.wpa_set_bss+0x170): undefined reference to `esp_wifi_get_config'

If its an order issue, one thing I've noted is that `esp-idf/esp_wifi/libesp_wifi.a` and `esp-idf/wpa_supplicant/libwpa_supplicant.a` is include far before `/Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/esp_wifi/lib/esp32c6/libnet80211.a` which has `esp_wifi_get_config` declared in it, but its also included 6x in the same linking, but its always much after `esp-idf/esp_wifi/libesp_wifi.a` and `esp-idf/wpa_supplicant/libwpa_supplicant.a`

```bash
: && /Users/josephmontanez_1/.espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/bin/riscv32-esp-elf-g++ -march=rv32imac_zicsr_zifencei -nostartfiles -march=rv32imac_zicsr_zifencei --specs=nosys.specs    -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32C6=0 -Wl,--Map=/Users/josephmontanez_1/Documents/zig-esp-idf-sample/build/zig-sample-idf.map -Wl,--no-warn-rwx-segments -Wl,--orphan-handling=warn -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T rom.api.ld -T esp32c6.peripherals.ld -T esp32c6.rom.ld -T esp32c6.rom.api.ld -T esp32c6.rom.rvfp.ld -T esp32c6.rom.wdt.ld -T esp32c6.rom.systimer.ld -T esp32c6.rom.version.ld -T esp32c6.rom.phy.ld -T esp32c6.rom.coexist.ld -T esp32c6.rom.net80211.ld -T esp32c6.rom.pp.ld -T esp32c6.rom.newlib.ld -T esp32c6.rom.newlib-normal.ld -T esp32c6.rom.heap.ld -T memory.ld -T sections.ld CMakeFiles/zig-sample-idf.elf.dir/project_elf_src_esp32c6.c.obj -o zig-sample-idf.elf -L/Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/riscv/ld   -L/Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/soc/esp32c6/ld   -L/Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/esp_rom/esp32c6/ld   -L/Users/josephmontanez_1/Documents/zig-esp-idf-sample/build/esp-idf/esp_system/ld   -L/Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/esp_phy/lib/esp32c6   -L/Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/esp_coex/lib/esp32c6   -L/Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/esp_wifi/lib/esp32c6 esp-idf/riscv/libriscv.a  esp-idf/esp_driver_gpio/libesp_driver_gpio.a  esp-idf/esp_pm/libesp_pm.a  esp-idf/mbedtls/libmbedtls.a  esp-idf/esp_app_format/libesp_app_format.a  esp-idf/esp_bootloader_format/libesp_bootloader_format.a  esp-idf/app_update/libapp_update.a  esp-idf/esp_partition/libesp_partition.a  esp-idf/efuse/libefuse.a  esp-idf/bootloader_support/libbootloader_support.a  esp-idf/esp_mm/libesp_mm.a  esp-idf/spi_flash/libspi_flash.a  esp-idf/esp_system/libesp_system.a  esp-idf/esp_common/libesp_common.a  esp-idf/esp_rom/libesp_rom.a  esp-idf/hal/libhal.a  esp-idf/log/liblog.a  esp-idf/heap/libheap.a  esp-idf/soc/libsoc.a  esp-idf/esp_security/libesp_security.a  esp-idf/esp_hw_support/libesp_hw_support.a  esp-idf/freertos/libfreertos.a  esp-idf/newlib/libnewlib.a  esp-idf/pthread/libpthread.a  esp-idf/cxx/libcxx.a  esp-idf/esp_timer/libesp_timer.a  esp-idf/esp_driver_gptimer/libesp_driver_gptimer.a  esp-idf/esp_ringbuf/libesp_ringbuf.a  esp-idf/esp_driver_uart/libesp_driver_uart.a  esp-idf/app_trace/libapp_trace.a  esp-idf/esp_event/libesp_event.a  esp-idf/nvs_flash/libnvs_flash.a  esp-idf/esp_driver_pcnt/libesp_driver_pcnt.a  esp-idf/esp_driver_spi/libesp_driver_spi.a  esp-idf/esp_driver_mcpwm/libesp_driver_mcpwm.a  esp-idf/esp_driver_i2s/libesp_driver_i2s.a  esp-idf/sdmmc/libsdmmc.a  esp-idf/esp_driver_sdspi/libesp_driver_sdspi.a  esp-idf/esp_driver_sdio/libesp_driver_sdio.a  esp-idf/esp_driver_rmt/libesp_driver_rmt.a  esp-idf/esp_driver_tsens/libesp_driver_tsens.a  esp-idf/esp_driver_sdm/libesp_driver_sdm.a  esp-idf/esp_driver_i2c/libesp_driver_i2c.a  esp-idf/esp_driver_ledc/libesp_driver_ledc.a  esp-idf/esp_driver_parlio/libesp_driver_parlio.a  esp-idf/esp_driver_usb_serial_jtag/libesp_driver_usb_serial_jtag.a  esp-idf/driver/libdriver.a  esp-idf/esp_phy/libesp_phy.a  esp-idf/esp_vfs_console/libesp_vfs_console.a  esp-idf/vfs/libvfs.a  esp-idf/lwip/liblwip.a  esp-idf/esp_netif/libesp_netif.a  esp-idf/wpa_supplicant/libwpa_supplicant.a  esp-idf/esp_coex/libesp_coex.a  esp-idf/esp_wifi/libesp_wifi.a  esp-idf/unity/libunity.a  esp-idf/cmock/libcmock.a  esp-idf/console/libconsole.a  esp-idf/http_parser/libhttp_parser.a  esp-idf/esp-tls/libesp-tls.a  esp-idf/esp_adc/libesp_adc.a  esp-idf/esp_driver_cam/libesp_driver_cam.a  esp-idf/esp_eth/libesp_eth.a  esp-idf/esp_gdbstub/libesp_gdbstub.a  esp-idf/esp_hid/libesp_hid.a  esp-idf/tcp_transport/libtcp_transport.a  esp-idf/esp_http_client/libesp_http_client.a  esp-idf/esp_http_server/libesp_http_server.a  esp-idf/esp_https_ota/libesp_https_ota.a  esp-idf/esp_https_server/libesp_https_server.a  esp-idf/esp_lcd/libesp_lcd.a  esp-idf/protobuf-c/libprotobuf-c.a  esp-idf/protocomm/libprotocomm.a  esp-idf/esp_local_ctrl/libesp_local_ctrl.a  esp-idf/espcoredump/libespcoredump.a  esp-idf/wear_levelling/libwear_levelling.a  esp-idf/fatfs/libfatfs.a  esp-idf/ieee802154/libieee802154.a  esp-idf/json/libjson.a  esp-idf/mqtt/libmqtt.a  esp-idf/nvs_sec_provider/libnvs_sec_provider.a  esp-idf/rt/librt.a  esp-idf/spiffs/libspiffs.a  esp-idf/wifi_provisioning/libwifi_provisioning.a  esp-idf/main/libmain.a  esp-idf/app_trace/libapp_tr [... cut ...]
│ libapp_update.a                 │        232 │     4 │    4 │     0 │     0 │        228 │      30 │        0 │    198 │
│ libstdc++.a                     │         99 │     5 │    5 │     0 │     0 │         94 │       4 │        0 │     90 │
│ libcxx.a                        │         16 │     0 │    0 │     0 │     0 │         16 │       0 │        0 │     16 │
│ libnvs_sec_provider.a           │          2 │     0 │    0 │     0 │     0 │          2 │       0 │        0 │      2 │
└─────────────────────────────────┴────────────┴───────┴──────┴───────┴───────┴────────────┴─────────┴──────────┴────────┘
```

kassane, 2025-02-03:
> However I also need to comment out some functions as these are defined in `libnet80211.a`

Also related (esp32-s3):
- https://github.com/kassane/zig-esp-idf-sample/discussions/12#discussioncomment-10367911

joseph-montanez, 2025-02-04:
Minor update on my end:

**cmake/zig-build.cmake**
```zig
pub fn searched_idf_libs(b: *std.Build, lib: *std.Build.Step.Compile) !void {
    var dir = try std.fs.cwd().openDir("../build", .{
        .iterate = true,
    });
    defer dir.close();
    var walker = try dir.walk(b.allocator);
    defer walker.deinit();

    while (try walker.next()) |entry| {
        const ext = std.fs.path.extension(entry.basename);
        const lib_ext = inline for (&.{".obj"}) |e| {
            if (std.mem.eql(u8, ext, e))
                break true;
        } else false;
        if (lib_ext) {
            const src_path = std.fs.path.dirname(@src().file) orelse b.pathResolve(&.{".."});
            const cwd_path = b.pathJoin(&.{ src_path, "build", b.dupe(entry.path) });
            const lib_file: std.Build.LazyPath = .{ .cwd_relative = cwd_path };
            lib.addObjectFile(lib_file);
        }
    }
}
```

Two things here.  First, the lib even when compile ReleaseSmall is really large:
```
                                                        Per-archive contributions to ELF file                                                        
┏━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━┳━━━━━━━┳━━━━━━━┳━━━━━━━┳━━━━━━━┳━━━━━━━━━━━━┳━━━━━━━━━┳━━━━━━━━━━┳━━━━━━━━┳━━━━━━━━━┳━━━━━━━━━━━━━━━┓
┃ Archive File                    ┃ Total Size ┃ DIRAM ┃  .bss ┃ .data ┃ .text ┃ Flash Code ┃ .rodata ┃ .appdesc ┃  .text ┃ LP SRAM ┃ .rtc_reserved ┃
┡━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━╇━━━━━━━╇━━━━━━━╇━━━━━━━╇━━━━━━━╇━━━━━━━━━━━━╇━━━━━━━━━╇━━━━━━━━━━╇━━━━━━━━╇━━━━━━━━━╇━━━━━━━━━━━━━━━┩
│ libapp_zig.a                    │     279593 │  4935 │  4065 │   536 │   334 │     274658 │   13860 │        0 │ 260798 │       0 │             0 │
│ libnet80211.a                   │     185156 │ 19485 │ 10386 │  1785 │  7314 │     165671 │   18735 │        0 │ 146936 │       0 │             0 │
│ libpp.a                         │     110742 │ 45371 │  2746 │  3201 │ 39424 │      65371 │    9405 │        0 │  55966 │       0 │             0 │
│ libesp_app_format.a             │      59659 │    10 │    10 │     0 │     0 │      59649 │   58863 │      256 │    530 │       0 │             0 │
│ libesp_hw_support.a             │      42337 │ 12636 │   119 │  4347 │  8170 │      29677 │    2957 │        0 │  26720 │      24 │            24 │
│ libphy.a                        │      28375 │  5547 │    44 │   919 │  4584 │      22828 │       0 │        0 │  22828 │       0 │             0 │
│ libnvs_flash.a                  │      17022 │    28 │    28 │     0 │     0 │      16994 │     252 │        0 │  16742 │       0 │             0 │
│ libfreertos.a                   │      16319 │ 14982 │  2240 │    88 │ 12654 │       1337 │    1007 │        0 │    330 │       0 │             0 │
│ libhal.a                        │      13861 │  8606 │    12 │   348 │  8246 │       5255 │     347 │        0 │   4908 │       0 │             0 │
│ libesp_system.a                 │      13644 │  2664 │   138 │    92 │  2434 │      10980 │     570 │        0 │  10410 │       0 │             0 │
│ libspi_flash.a                  │      10558 │  8981 │    14 │   941 │  8026 │       1577 │     357 │        0 │   1220 │       0 │             0 │
│ libc.a                          │       8886 │   712 │   356 │   356 │     0 │       8174 │      96 │        0 │   8078 │       0 │             0 │
│ libesp_driver_uart.a            │       8576 │   156 │    28 │   128 │     0 │       8420 │     552 │        0 │   7868 │       0 │             0 │
│ libmbedcrypto.a                 │       7214 │   228 │    96 │    88 │    44 │       6986 │     126 │        0 │   6860 │       0 │             0 │
│ libvfs.a                        │       6170 │   236 │    44 │   192 │     0 │       5934 │     158 │        0 │   5776 │       0 │             0 │
│ libcoexist.a                    │       5474 │  1544 │     2 │  1446 │    96 │       3930 │      40 │        0 │   3890 │       0 │             0 │
│ libnewlib.a                     │       4071 │  2270 │   200 │   156 │  1914 │       1801 │     111 │        0 │   1690 │       0 │             0 │
│ libheap.a                       │       3949 │  1324 │     8 │    12 │  1304 │       2625 │     437 │        0 │   2188 │       0 │             0 │
│ libesp_mm.a                     │       3772 │   662 │    44 │    28 │   590 │       3110 │     166 │        0 │   2944 │       0 │             0 │
│ libesp_timer.a                  │       3460 │  1566 │    36 │    16 │  1514 │       1894 │     104 │        0 │   1790 │       0 │             0 │
│ libesp_phy.a                    │       3197 │   397 │    44 │     1 │   352 │       2800 │     284 │        0 │   2516 │       0 │             0 │
│ libesp_driver_usb_serial_jtag.a │       2521 │   167 │    17 │    48 │   102 │       2354 │     152 │        0 │   2202 │       0 │             0 │
│ libesp_partition.a              │       2214 │     8 │     8 │     0 │     0 │       2206 │     168 │        0 │   2038 │       0 │             0 │
│ libesp_driver_gpio.a            │       2145 │     0 │     0 │     0 │     0 │       2145 │     277 │        0 │   1868 │       0 │             0 │
│ liblog.a                        │       1966 │   602 │   272 │     8 │   322 │       1364 │      48 │        0 │   1316 │       0 │             0 │
│ libesp_common.a                 │       1816 │     0 │     0 │     0 │     0 │       1816 │    1758 │        0 │     58 │       0 │             0 │
[... cut ...]
    log               # ESP logging system (for g_log_level)
    nvs_flash         # Non-volatile storage (for g_misc_nvs)
    esp_netif         # ESP networking interface
    esp_event         # ESP event loop (for esp_event_post)
    lwip              # Lightweight TCP/IP stack
    esp_phy           # Physical layer for WiFi
    esp_pm            # Power management (needed for WiFi stability)
)
```

IDF is not tracking `libapp_zig.a` so the subsequent included libs that would be present are excluded because it sees they are not used.

joseph-montanez, 2025-02-04:
On more issue... The structs do not align. I looked at idf-sys.zig and it says xtensa was used, but I am using RISCV, so I don't know if it just one, or many that do not align. I don't see anywhere on how to recreate `idf-sys.zig`.

kassane, 2025-02-04:
> On more issue... The structs do not align. I looked at idf-sys.zig and it says xtensa was used, but I am using RISCV, so I don't know if it just one, or many that do not align. I don't see anywhere on how to recreate `idf-sys.zig`.

It was primarily based on xtensa-gcc. However, that `build.zig` is currently using gcc includes for target configuration.
https://github.com/kassane/zig-esp-idf-sample/blob/0085d8160adca587af512bb5d94955d9c79c2c27/build.zig#L55-L65
Although `idf-sys` has xtensa bindings, this doesn't mean that it's going to be used without calling it explicit.
Doing so might cause the gnu-linker compiler to fail.

kassane, 2025-02-04:
> Two things here. First, the lib even when compile ReleaseSmall is really large (even removed `-freference-trace`):

Does `-freference-trace` affect build size?

> The second item and why WIFI fails to compile is because the `searched_idf_libs` function is taking every object file but not linking any libs or additional object files from IDF's core components. This is why `libnet80211.a` and `libmesh.a` where missing symbols.

Make sense. But then, what would it take to be able to get objects from those libraries?
The attempt to bind static-lib to static-lib does not work, justifying the 'obj' capture.

joseph-montanez, 2025-02-04:
> Does `-freference-trace` affect build size?
>
>
> Make sense. But then, what would it take to be able to get objects from those libraries? The attempt to bind static-lib to static-lib does not work, [...]

This goes back to this. I've dropped the obj capture and just include the libs. They are in their respective .a files so including them is enough and you won't need a large binary anymore. The problem here is you need to know what to look for since IDF is not building the Zig library. 
```cmake
target_link_libraries(${COMPONENT_LIB} PRIVATE
    "-Wl,--start-group"
    ${IDF_PATH}/components/esp_phy/lib/esp32c6/libphy.a
    ${IDF_PATH}/components/esp_wifi/lib/esp32c6/libpp.a
    ${IDF_PATH}/components/esp_wifi/lib/esp32c6/libmesh.a
    ${IDF_PATH}/components/esp_wifi/lib/esp32c6/libnet80211.a
    ${IDF_PATH}/components/esp_wifi/lib/esp32c6/libcore.a
    ${CMAKE_BINARY_DIR}/esp-idf/log/liblog.a
    ${CMAKE_BINARY_DIR}/esp-idf/nvs_flash/libnvs_flash.a
    ${CMAKE_BINARY_DIR}/esp-idf/soc/libsoc.a
    #${CMAKE_BINARY_DIR}/esp-idf/soc/libesp_event.a
    ${CMAKE_BINARY_DIR}/esp-idf/esp_hw_support/libesp_hw_support.a
    ${CMAKE_BINARY_DIR}/esp-idf/hal/libhal.a
    ${CMAKE_BINARY_DIR}/esp-idf/mbedtls/mbedtls/library/libmbedcrypto.a
    ${CMAKE_BINARY_DIR}/esp-idf/freertos/libfreertos.a
    ${CMAKE_BINARY_DIR}/esp-idf/lwip/liblwip.a
    ${CMAKE_BINARY_DIR}/esp-idf/esp_wifi/libesp_wifi.a
    ${CMAKE_BINARY_DIR}/esp-idf/esp_netif/libesp_netif.a
    ${CMAKE_BINARY_DIR}/esp-idf/esp_event/libesp_event.a
    ${CMAKE_BINARY_DIR}/esp-idf/wpa_supplicant/libwpa_supplicant.a
    ${CMAKE_BINARY_DIR}/lib/libapp_zig.a
    "-Wl,--end-group"
)
```

This brings the zig library to only 6KB versus the 200KB+ (for ReleaseSmall):
```
                                                        Per-archive contributions to ELF file
┏━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━┳━━━━━━━━━━━━┳━━━━━━━┳━━━━━━━┳━━━━━━━┳━━━━━━━┳━━━━━━━━━━━━┳━━━━━━━━━┳━━━━━━━━━━┳━━━━━━━━┳━━━━━━━━━┳━━━━━━━━━━━━━━━┓
┃ Archive File                    ┃ Total Size ┃ DIRAM ┃  .bss ┃ .data ┃ .text ┃ Flash Code ┃ .rodata ┃ .appdesc ┃  .text ┃ LP SRAM ┃ .rtc_reserved ┃
┡━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━╇━━━━━━━━━━━━╇━━━━━━━╇━━━━━━━╇━━━━━━━╇━━━━━━━╇━━━━━━━━━━━━╇━━━━━━━━━╇━━━━━━━━━━╇━━━━━━━━╇━━━━━━━━━╇━━━━━━━━━━━━━━━┩
│ libnet80211.a                   │     184898 │ 19485 │ 10386 │  1785 │  7314 │     165413 │   18735 │        0 │ 146678 │       0 │             0 │
│ libpp.a                         │     110746 │ 45365 │  2746 │  3201 │ 39418 │      65381 │    9405 │        0 │  55976 │       0 │             0 │
│ liblwip.a                       │     102651 │  2482 │  2470 │    12 │     0 │     100169 │    3455 │        0 │  96714 │       0 │             0 │
│ libmbedcrypto.a                 │      79609 │   454 │   252 │   116 │    86 │      79155 │    6569 │        0 │  72586 │       0 │             0 │
│ libwpa_supplicant.a             │      78621 │  1329 │  1321 │     8 │     0 │      77292 │    1596 │        0 │  75696 │       0 │             0 │
│ libesp_app_format.a             │      59659 │    10 │    10 │     0 │     0 │      59649 │   58863 │      256 │    530 │       0 │             0 │
│ libesp_hw_support.a             │      42337 │ 12636 │   119 │  4347 │  8170 │      29677 │    2957 │        0 │  26720 │      24 │            24 │
│ libphy.a                        │      28375 │  5547 │    44 │   919 │  4584 │      22828 │       0 │        0 │  22828 │       0 │             0 │
│ libfreertos.a                   │      17343 │ 15908 │  2240 │    88 │ 13580 │       1435 │    1105 │        0 │    330 │       0 │             0 │
│ libnvs_flash.a                  │      17022 │    28 │    28 │     0 │     0 │      16994 │     252 │        0 │  16742 │       0 │             0 │
│ libhal.a                        │      15097 │  8606 │    12 │   348 │  8246 │       6491 │     399 │        0 │   6092 │       0 │             0 │
│ libesp_system.a                 │      13644 │  2664 │   138 │    92 │  2434 │      10980 │     570 │        0 │  10410 │       0 │             0 │
│ libspi_flash.a                  │      10558 │  8981 │    14 │   941 │  8026 │       1577 │     357 │        0 │   1220 │       0 │             0 │
│ libc.a                          │       8886 │   712 │   356 │   356 │     0 │       8174 │      96 │        0 │   8078 │       0 │             0 │
│ libesp_driver_uart.a            │       8576 │   156 │    28 │   128 │     0 │       8420 │     552 │        0 │   7868 │       0 │             0 │
│ libesp_netif.a                  │       8210 │    33 │    29 │     4 │     0 │       8177 │     163 │        0 │   8014 │       0 │             0 │
│ libvfs.a                        │       6170 │   236 │    44 │   192 │     0 │       5934 │     158 │        0 │   5776 │       0 │             0 │
│ libapp_zig.a                    │       6162 │    20 │    20 │     0 │     0 │       6142 │    1842 │        0 │   4300 │       0 │             0 │
│ libcoexist.a                    │       5474 │  1544 │     2 │  1446 │    96 │       3930 │      40 │        0 │   3890 │       0 │             0 │
│ libesp_wifi.a                   │       5113 │   919 │    43 │   484 │   392 │       4194 │      42 │        0 │   4152 │       0 │             0 │
│ libnewlib.a                     │       4071 │  2270 │   200 │   156 │  1914 │       1801 │     111 │        0 │   1690 │       0 │             0 │
[... cut ...]
│ libmesh.a                       │         68 │     4 │     4 │     0 │     0 │         64 │       0 │        0 │     64 │       0 │             0 │
│ libcxx.a                        │         60 │     0 │     0 │     0 │     0 │         60 │       0 │        0 │     60 │       0 │             0 │
│ libnvs_sec_provider.a           │          2 │     0 │     0 │     0 │     0 │          2 │       0 │        0 │      2 │       0 │             0 │
└─────────────────────────────────┴────────────┴───────┴───────┴───────┴───────┴────────────┴─────────┴──────────┴────────┴─────────┴───────────────┘
```

joseph-montanez, 2025-02-05:
I got it working, regenerating the sys-idf.zig worked. 

I (773) wifi:connected with XXXXX XXXXXX, aid = 1, channel 1, BW20, bssid = XX:XX:XX:XX:XX:XX
I (773) wifi:security: WPA2-PSK, phy:11bgn, rssi:-36, cipher(pairwise:0x3, group:0x3), pmf:0,
I (793) wifi:pm start, type: 1, twt_start:0

I (793) wifi:pm start, type:1, aid:0x1, trans-BSSID:XX:XX:XX:XX:XX:XX, BSSID[5]:0xXX, mbssid(max-indicator:0, index:0), he:0
I (803) wifi:dp: 1, bi: 102400, li: 3, scale listen interval from 307200 us to 307200 us
I (803) wifi:set rx beacon pti, rx_bcn_pti: 10, bcn_timeout: 25000, mt_pti: 10, mt_time: 10000
I (813) wifi:[ADDBA]TX addba request, tid:0, dialogtoken:1, bufsize:64, A-MSDU:0(not supported), policy:1(IMR), ssn:0(0x0)
I (823) wifi:[ADDBA]TX addba request, tid:7, dialogtoken:2, bufsize:64, A-MSDU:0(not supported), policy:1(IMR), ssn:0(0x20)
I (833) wifi:[ADDBA]TX addba request, tid:5, dialogtoken:3, bufsize:64, A-MSDU:0(not supported), policy:1(IMR), ssn:0(0x0)
I (843) wifi:AP's beacon interval = 102400 us, DTIM period = 1
I (853) wifi:[ADDBA]RX addba response, status:0, tid:0/tb:1(0xa1), bufsize:64, batimeout:0, txa_wnd:64
I (863) wifi:[ADDBA]RX addba response, status:0, tid:7/tb:1(0xa1), bufsize:64, batimeout:0, txa_wnd:64
I (873) wifi:[ADDBA]RX addba response, status:0, tid:5/tb:1(0xa1), bufsize:64, batimeout:0, txa_wnd:64
I (1903) esp_netif_handlers: sta ip: 192.168.3.143, mask: 255.255.255.0, gw: 192.168.3.1

Here is what I did for my version:
```bash
# How to find the include paths
# find ~/Downloads/esp-idf-v5.4 ~/Documents/esp-wifi ~/.espressif -name "sys_arch.h"
# grep -R "rv_utils_get_core_id" ~/Downloads/esp-idf-v5.4 ~/Documents/esp-wifi

zig translate-c \
    -D target=riscv32-freestanding-none \
    -D cpu_generic_rv32+a+c+m+zicsr+zifencei \
    -D CONFIG_IDF_TARGET_ESP32C6 \
    -D __riscv \
    -D __COUNTER__=0 \
    -D IRAM_ATTR= \
    -D "_SECTION_ATTR_IMPL(x,y)"= \
    -D SOC_MMU_PAGE_SIZE=0x8000 \
    -I /Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/freertos/FreeRTOS-Kernel/include \
    -I /Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/freertos/config/include/freertos/ \
    -I /Users/josephmontanez_1/Documents/esp-wifi/build/config \
    -I /Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/freertos/config/riscv/include \
    -I /Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/freertos/FreeRTOS-Kernel-SMP/portable/riscv/include/freertos \
    -I /Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/esp_hw_support/include \
    -I /Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/soc/esp32c6/include \
    -I /Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/esp_common/include \
    -I /Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/riscv/include \
    -I /Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/soc/esp32c6/register \
    -I /Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/esp_system/include \
    -I /Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/newlib/include \
    -I /Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/newlib/platform_include/sys \
    -I /Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/newlib/platform_include \
    -I /Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/hal/platform_port/include \
    -I /Users/josephmontanez_1/.espressif/tools/riscv32-esp-elf/sys-include \
    -I /Users/josephmontanez_1/.espressif/tools/riscv32-esp-elf/esp-14.2.0_20241119/riscv32-esp-elf/riscv32-esp-elf/include \
    -I /Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/heap/include \
    -I /Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/esp_rom/include \
    -I /Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/esp_wifi/include \
    -I /Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/esp_event/include \
    -I /Users/josephmontanez_1/Downloads/esp-idf-v5.4/examples/protocols/linux_stubs/esp_stubs/include \
    -I /Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/lwip/include \
    -I /Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/lwip/lwip/src/include \
    -I /Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/lwip/port/include \
    -I /Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/freertos/config/include \
    -I /Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/lwip/port/freertos/include \
    -I /Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/lwip/port/esp32xx/include \
    -I /Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/log/include \
    -I /Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/nvs_flash/include \
    -I /Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/esp_partition/include \
    -I /Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/esp_netif/include \
    -I /Users/josephmontanez_1/Downloads/esp-idf-v5.4/components/esp_event/include \
    ~/Documents/esp-wifi/main/station_example_main.c > idf-sys.zig
```

The wifi C code is from the example from ESP-IDF. This changed all errors to be in global scope, so I had to change the code to work with that instead of ENUMs. I had to remove **app_main** from my generated version of `idf-sys.zig`, and change back the wifi sta opaque. Technically the wifi sat can stay opaque, but I just wanted to get this working first. Once I do a little code clean up, I'll push up my repository with the changes. I don't think anything I did is reusable though. idf-sys depends on the person's device and while `@cImport` is ideal, noting how many includes are needed and the following defines might be impractical.

kassane, 2025-02-05:
> I got it working, regenerating the sys-idf.zig worked.

@joseph-montanez, awesome!
PR's welcome.