kassane/zig-esp-idf-sample #25: `@cImport` no detect esp-idf components include

kassane, 2025-02-07:
> [!NOTE]
> Moving the zig-upstream discussion to an issue in the context of supporting espressif tools.

From @wiiggee1,

> I've looked at it, and I get it to compile. But it doesn't seem to find the c header files.

esp-idf does not provide an environment variable to access include components, but it is possible via CMake in this way.

https://github.com/kassane/zig-esp-idf-sample/blob/21977c7b438fc59acb2e0203d0949b9396879d1a/cmake/zig-build.cmake#L89
https://github.com/kassane/zig-esp-idf-sample/blob/21977c7b438fc59acb2e0203d0949b9396879d1a/build.zig#L24

In this project for example **translate-c** is used to generate `idf-sys.zig` (low-level API).
Based on https://github.com/kassane/zig-esp-idf-sample/blob/main/docs/build-internals.md#buildzig-details

Or you can generate your own bindgen similar to https://github.com/kassane/zig-esp-idf-sample/issues/22#issuecomment-2636196075 .

wiiggee1, 2025-02-07:
This is my current `CMakeLists.txt` and my `build.zig`: 

```
set(REQ_DEPENDENCIES freertos) 

idf_component_register(
    SRCS "linker_placeholder.c"
    INCLUDE_DIRS "."
    PRIV_REQUIRES ${REQ_DEPENDENCIES} 
)

set(include_dirs $<TARGET_PROPERTY:${COMPONENT_LIB},INCLUDE_DIRECTORIES> ${CMAKE_C_IMPLICIT_INCLUDE_DIRECTORIES})

# Create a custom target that generates an expor

message(STATUS "CMAKE_CURRENT_BINARY_DIR: ${CMAKE_CURRENT_BINARY_DIR}")
message(STATUS "CMAKE_CURRENT_SOURCE_DIR: ${CMAKE_CURRENT_SOURCE_DIR}")

add_prebuilt_library(zig_main ${CMAKE_CURRENT_SOURCE_DIR}/zig-out/lib/libzig_main.a)
#add_prebuilt_library(zig_main ${CMAKE_CURRENT_BINARY_DIR}/lib/libzig_main.a)

# add_dependencies(zig_main zig_build)

# main calls these functions that is defined from the static library, so link to main. 

# target_link_libraries(${COMPONENT_LIB} PUBLIC  $<TARGET_OBJECTS:zig_main>)
target_link_libraries(${COMPONENT_LIB} PRIVATE zig_main )

#target_link_libraries(${COMPONENT_LIB} PRIVATE ${CMAKE_CURRENT_SOURCE_DIR}/zig-out/lib/libzig_main.a)

message(STATUS "Component lib env: ${COMPONENT_LIB}")
message(STATUS "INCLUDE_DIRS env: ${INCLUDE_DIRS}")

```

```zig
const std = @import("std");

// Although this function looks imperative, note that its job is to
// declaratively construct a build graph that will be executed by an external
// runner.
pub fn build(b: *std.Build) void {
    // Standard target options allows the person running `zig build` to choose
    // what target to build for. Here we do not override the defaults, which
    // means any target is allowed, and the default is native. Other options
    // for restricting supported target set are available.
    const esp32s3_target = std.Target.Query{
        .cpu_arch = .xtensa,
        .cpu_model = .{ .explicit = &std.Target.xtensa.cpu.esp32s3 },
        .os_tag = .freestanding,
        .abi = .none,
    };

    const target = b.resolveTargetQuery(esp32s3_target);
    const optimize = b.standardOptimizeOption(.{ .preferred_optimize_mode = .ReleaseSafe });

    // Ensure that Zig can find the necessary ESP-IDF header files.
    const esp_idf_path = std.process.getEnvVarOwned(b.allocator, "IDF_PATH") catch "/home/wiiggee1/esp/esp-idf/";
    std.debug.print("esp_idf_path: {s}\n", .{esp_idf_path});

    const esp_components = std.fmt.allocPrint(b.allocator, "{s}/components", .{esp_idf_path}) catch @panic("Out of Memory");
    //const freertos = std.fmt.allocPrint(b.allocator, "{s}/components/freertos/FreeRTOS-Kernel/include/freertos", .{esp_idf_path}) catch @panic("Out of Memory");

    const zig_main_exe = b.addExecutable(.{
        .name = "zig_main_exe",
        .root_source_file = b.path("src/main.zig"),
        .target = target,
        .optimize = optimize,
    });

[... 101 lines cut ...]
    // const lib_unit_tests = b.addTest(.{
    //     .root_source_file = b.path("src/root.zig"),
    //     .target = target,
    //     .optimize = optimize,
    // });

    //const run_lib_unit_tests = b.addRunArtifact(lib_unit_tests);

    // const exe_unit_tests = b.addTest(.{
    //     .root_source_file = b.path("src/main.zig"),
    //     .target = target,
    //     .optimize = optimize,
    // });

    // const run_exe_unit_tests = b.addRunArtifact(exe_unit_tests);

    // Similar to creating the run step earlier, this exposes a `test` step to
    // the `zig build --help` menu, providing a way for the user to request
    // running the unit tests.
    // const test_step = b.step("test", "Run unit tests");
    //test_step.dependOn(&run_lib_unit_tests.step);
    // test_step.dependOn(&run_exe_unit_tests.step);
}

```

kassane, 2025-02-08:
<details>
<summary>main.zig</summary>

```zig
const std = @import("std");
const c = @cImport({
    // @cInclude("esp_log.h");
    @cInclude("freertos/FreeRTOS.h");
    @cInclude("freertos/task.h");
});

export fn app_main() callconv(.C) void {
    // _ = std.c.printf("Hello from Zig!\n");
    // c.esp_log_write(c.ESP_LOG_INFO, "main", "Hello from Zig!\n");
    c.vTaskDelay(1000);
}
```
</details>

I briefly tested your solution and got this error.
```bash
src/main.zig:2:11: error: C import failed
const c = @cImport({
          ^~~~~~~~
referenced by:
    app_main: src/main.zig:11:5
    app_main: src/main.zig:8:1
    4 reference(s) hidden; use '-freference-trace=6' to see all references
/home/kassane/zig-relsafe-espressif-x86_64-linux-musl-baseline/lib/include/inttypes.h:24:15: error: 'inttypes.h' file not found
#include_next <inttypes.h>
              ^
```
However, when I added the `espressif-gcc` (sys-)include get:
https://github.com/kassane/zig-esp-idf-sample/blob/21977c7b438fc59acb2e0203d0949b9396879d1a/build.zig#L55-L66
```bash
src/main.zig:2:11: error: C import failed
const c = @cImport({
          ^~~~~~~~
referenced by:
    app_main: src/main.zig:11:5
    app_main: src/main.zig:8:1
    4 reference(s) hidden; use '-freference-trace=6' to see all references
/home/kassane/zigEsp32s3/.zig-cache/o/9affb8fb49dba6c9509d22e244c3e7a3/cimport.h:1:10: error: 'freertos/FreeRTOS.h' file not found
#include <freertos/FreeRTOS.h>
         ^
```

wiiggee1, 2025-02-08:
@kassane Hi again this is the error I get: 

```
src/main.zig:5:18: error: C import failed
const freertos = @cImport({
                 ^~~~~~~~
referenced by:
    app_main: src/main.zig:20:5
    app_main: src/main.zig:11:1
    root: /home/wiiggee1/zig_versions/zig-relsafe-espressif-x86_64-linux-musl-baseline/lib/std/start.zig:3:22
    comptime: /home/wiiggee1/zig_versions/zig-relsafe-espressif-x86_64-linux-musl-baseline/lib/std/start.zig:27:9
    start: /home/wiiggee1/zig_versions/zig-relsafe-espressif-x86_64-linux-musl-baseline/lib/std/std.zig:96:27
    comptime: /home/wiiggee1/zig_versions/zig-relsafe-espressif-x86_64-linux-musl-baseline/lib/std/std.zig:160:9
/home/wiiggee1/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:63:10: error: 'FreeRTOSConfig.h' file not found
#include "FreeRTOSConfig.h"
         ^
error: the following command failed with 2 compilation errors:
/home/wiiggee1/zig_versions/zig-relsafe-espressif-x86_64-linux-musl-baseline/zig build-lib -freference-trace=6 -ODebug -target xtensa-freestanding-none -mcpu esp32s3 -I /home/wiiggee1/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/xtensa-esp-elf/include -Mroot=/home/wiiggee1/Desktop/Master_Thesis/MasterThesis_EdgeAI/main/src/main.zig -lc --cache-dir /home/wiiggee1/Desktop/Master_Thesis/MasterThesis_EdgeAI/main/.zig-cache --global-cache-dir /home/wiiggee1/.cache/zig --name zig_main -static --script /home/wiiggee1/Desktop/Master_Thesis/MasterThesis_EdgeAI/main/linker.ld --zig-lib-dir /home/wiiggee1/zig_versions/zig-relsafe-espressif-x86_64-linux-musl-baseline/lib/ --listen=-
Build Summary: 0/3 steps succeeded; 1 failed
install transitive failure
└─ install zig_main transitive failure
   └─ zig build-lib zig_main Debug xtensa-freestanding-none 2 errors
error: the following build command failed with exit code 1:
/home/wiiggee1/Desktop/Master_Thesis/MasterThesis_EdgeAI/main/.zig-cache/o/b5e9fe4e3f732e1322ec71da4132cea4/build /home/wiiggee1/zig_versions/zig-relsafe-espressif-x86_64-linux-musl-baseline/zig /home/wiiggee1/zig_versions/zig-relsafe-espressif-x86_64-linux-musl-baseline/lib /home/wiiggee1/Desktop/Master_Thesis/MasterThesis_EdgeAI/main /home/wiiggee1/Desktop/Master_Thesis/MasterThesis_EdgeAI/main/.zig-cache /home/wiiggee1/.cache/zig --seed 0x63f792f0 -Z872249f329b08350 -freference-trace=6
```

wiiggee1, 2025-02-08:
For example, here I can find functions related to the `hal_utils.h`header file. 
[image]

While for other such as `freertos.h`, I cant seem to find the header file: 

[image]

Below is my `build.zig`: 

```zig
const std = @import("std");

// Although this function looks imperative, note that its job is to
// declaratively construct a build graph that will be executed by an external
// runner.
pub fn build(b: *std.Build) !void {
    // Standard target options allows the person running `zig build` to choose
    // what target to build for. Here we do not override the defaults, which
    // means any target is allowed, and the default is native. Other options
    // for restricting supported target set are available.
    const esp32s3_target = std.Target.Query{
        .cpu_arch = .xtensa,
        .cpu_model = .{ .explicit = &std.Target.xtensa.cpu.esp32s3 },
        .os_tag = .freestanding,
        .abi = .none,
    };

    const target = b.resolveTargetQuery(esp32s3_target);
    const optimize = b.standardOptimizeOption(.{});

    // Ensure that Zig can find the necessary ESP-IDF header files.
    const home_directory = std.process.getEnvVarOwned(b.allocator, "HOME") catch "";
    const esp_idf_path = std.process.getEnvVarOwned(b.allocator, "IDF_PATH") catch "/home/wiiggee1/esp/esp-idf/";
    std.debug.print("esp_idf_path: {s}\n", .{esp_idf_path});

    //Here goes a static library to be linked with c / c++ app.
    const esp_idf_lib = b.addStaticLibrary(.{
        .name = "zig_main",
        .root_source_file = b.path("src/main.zig"),
        .target = target,
        .optimize = optimize,
    });

    if (!std.mem.eql(u8, home_directory, "")) {
        const archtools = b.fmt("{s}-esp-elf", .{
            @tagName(esp_idf_lib.rootModuleTarget().cpu.arch),
        });

        esp_idf_lib.addIncludePath(.{ .cwd_relative = b.pathJoin(&.{
            home_directory,
            ".espressif",
            "tools",
            archtools,
            "esp-14.2.0_20241119",
            archtools,
            archtools,
            "include",
        }) });
    }

    if (!std.mem.eql(u8, esp_idf_path, "")) {
        try searched_idf_include(b, esp_idf_lib, esp_idf_path);
        //try add_c_includes(b, esp_idf_lib);
        esp_idf_lib.linkLibC();
    }

    b.installArtifact(esp_idf_lib);

    esp_idf_lib.setLinkerScript(b.path("linker.ld"));

[... 97 lines cut ...]

pub fn searched_idf_include(b: *std.Build, lib: *std.Build.Step.Compile, idf_path: []const u8) !void {
    const comp = b.pathJoin(&.{ idf_path, "components" });
    var dir = try std.fs.cwd().openDir(comp, .{
        .iterate = true,
    });
    defer dir.close();
    var walker = try dir.walk(b.allocator);
    defer walker.deinit();

    while (try walker.next()) |entry| {
        const ext = std.fs.path.extension(entry.basename);
        const include_file = inline for (&.{".h"}) |e| {
            if (std.mem.eql(u8, ext, e))
                break true;
        } else false;
        if (include_file) {
            const include_dir = b.pathJoin(&.{ comp, std.fs.path.dirname(b.dupe(entry.path)).? });
            std.debug.print("component include dir: {s}", .{include_dir});
            _ = lib.out_filename;
            // lib.addIncludePath(.{ .cwd_relative = include_dir });
        }
    }
}
```

kassane, 2025-02-08:
Currently `build.zig` get only components main-include, like: `-isystem /home/kassane/esp/v5.4/esp-idf/components/freertos/include` - missing another subdirs.
```bash
$ find /home/kassane/esp/v5.4/esp-idf/components/freertos -type d -name "include"
/home/kassane/esp/v5.4/esp-idf/components/freertos/FreeRTOS-Kernel/include
/home/kassane/esp/v5.4/esp-idf/components/freertos/FreeRTOS-Kernel/portable/linux/include
/home/kassane/esp/v5.4/esp-idf/components/freertos/FreeRTOS-Kernel/portable/riscv/include
/home/kassane/esp/v5.4/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include
/home/kassane/esp/v5.4/esp-idf/components/freertos/FreeRTOS-Kernel-SMP/include
/home/kassane/esp/v5.4/esp-idf/components/freertos/FreeRTOS-Kernel-SMP/portable/linux/include
/home/kassane/esp/v5.4/esp-idf/components/freertos/FreeRTOS-Kernel-SMP/portable/riscv/include
/home/kassane/esp/v5.4/esp-idf/components/freertos/FreeRTOS-Kernel-SMP/portable/xtensa/include
/home/kassane/esp/v5.4/esp-idf/components/freertos/config/include
/home/kassane/esp/v5.4/esp-idf/components/freertos/config/linux/include
/home/kassane/esp/v5.4/esp-idf/components/freertos/config/riscv/include
/home/kassane/esp/v5.4/esp-idf/components/freertos/config/xtensa/include
/home/kassane/esp/v5.4/esp-idf/components/freertos/esp_additions/include
```

wiiggee1, 2025-02-08:
@kassane Can you elaborate what you mean. Should I change to `addSystemIncludePath`instead. Or what part am I missing to include?

kassane, 2025-02-08:
> Or what part am I missing to include?

Sorry for my poor explanation.

Just add the remaining includes and fix current freertos includepath.
```diff
- freertos/include
+ freertos/FreeRTOS-Kernel/include
+ freertos/config/include
```

The idea of `searched_idf_include` is to track all includes available in the rootpath.

kassane, 2025-02-08:
Try generate `bindings.zig` - need esp32s3 `sdkconfig.h`:

<details>
<summary>bindgen.sh</summary>

```bash
#!/usr/bin/env bash

ZIG_PATH=$HOME/zig-bootstrap/zig-espressif-bootstrap/out/zig-relsafe-espressif-x86_64-linux-musl-baseline

# "-fp-s32c1i" is the only way to skip floating point support
# https://github.com/espressif/llvm-project/issues/91

$ZIG_PATH/zig translate-c \
    -lc \
    -target xtensa-freestanding-none \
    -mcpu=esp32s3-fp-s32c1i \
    -D __xtensa \
    -D __COUNTER__=0 \
    -I $IDF_PATH/components/freertos/FreeRTOS-Kernel/include \
    -I $IDF_PATH/components/freertos/config/include/freertos/ \
    -I $HOME/Documents/esp-wifi/build/config \
    -I $IDF_PATH/components/freertos/config/xtensa/include \
    -I $IDF_PATH/components/freertos/FreeRTOS-Kernel-SMP/portable/xtensa/include/freertos \
    -I $IDF_PATH/components/esp_hw_support/include \
    -I $IDF_PATH/components/soc/esp32s3/include \
    -I $IDF_PATH/components/esp_common/include \
    -I $IDF_PATH/components/xtensa/include \
    -I $IDF_PATH/components/xtensa/esp32s3/include \
    -I $IDF_PATH/components/soc/esp32s3/register \
    -I $IDF_PATH/components/esp_system/include \
    -I $IDF_PATH/components/newlib/include \
    -I $IDF_PATH/components/newlib/platform_include/sys \
    -I $IDF_PATH/components/newlib/platform_include \
    -I $IDF_PATH/components/hal/platform_port/include \
    -I $HOME/.espressif/tools/xtensa-esp-elf/sys-include \
    -I $HOME/.espressif/tools/xtensa-esp-elf/esp-14.2.0_20241119/xtensa-esp-elf/xtensa-esp-elf/include \
    -I $IDF_PATH/components/heap/include \
    -I $IDF_PATH/components/esp_rom/include \
    -I $IDF_PATH/components/esp_wifi/include \
    -I $IDF_PATH/components/esp_event/include \
    -I $IDF_PATH/examples/protocols/linux_stubs/esp_stubs/include \
    -I $IDF_PATH/components/lwip/include \
    -I $IDF_PATH/components/lwip/lwip/src/include \
    -I $IDF_PATH/components/lwip/port/include \
    -I $IDF_PATH/components/freertos/config/include \
    -I $IDF_PATH/components/lwip/port/freertos/include \
    -I $IDF_PATH/components/lwip/port/esp32xx/include \
    -I $IDF_PATH/components/log/include \
    -I $IDF_PATH/components/nvs_flash/include \
    -I $IDF_PATH/components/esp_partition/include \
    -I $IDF_PATH/components/esp_netif/include \
    -I $IDF_PATH/components/esp_event/include \
    -I $PWD \
    bindings.h >  bindings.zig
```
</details>

Works!

But,

<details>
<summary>b.addTranslateC</summary>

It will replace `@cImport`/`@cInclude`, making bindings generate only via the build-system.
see: #13

```zig
    // generate bindings for the ESP-IDF library
    const bindgen = b.addTranslateC(.{
        .link_libc = true,
        .optimize = optimize,
        .target = target,
        .root_source_file = b.path("bindings.h"),
    });
    bindgen.defineCMacro("CONFIG_IDF_TARGET_ESP32S3", "1");
   
    // [...]    
    try searched_idf_include(b, bindgen, esp_idf_path);

    bindgen.addIncludePath(b.path(".")); // sdkconfig.h path
    // replace '@cImport' to const c = '@import("idf")'
    zig_main_exe.root_module.addImport("idf", bindgen.addModule("idf"));
```
</details>

emit:

```bash
install transitive failure
└─ install zig_main_exe transitive failure
   └─ zig build-exe zig_main_exe Debug xtensa-freestanding-none transitive failure
      └─ translate-c 1 errors
/home/kassane/esp/v5.4/esp-idf/components/esp_rom/include/esp_rom_sys.h:12:10: error: 'soc/reset_reasons.h' file not found
#include "soc/reset_reasons.h"
         ^
```

kassane, 2025-02-08:
Some changes in `searched_idf_include`, get:

**Total**
-  416 include paths in idf-components
-  20140 paths in idf-components

<details>
<summary>Draft  - build.zig</summary>

```zig
pub fn searched_idf_include(b: *std.Build, lib: *std.Build.Step.TranslateC, idf_path: []const u8) !void {
    // Build the components base directory.
    const comp = b.pathJoin(&.{ idf_path, "components" });
    var dir = try std.fs.cwd().openDir(comp, .{ .iterate = true });
    defer dir.close();
    var walker = try dir.walk(b.allocator);
    defer walker.deinit();

    // Use a hash map to track include directories already added.
    var added = std.StringHashMap(void).init(b.allocator);
    defer added.deinit();

    var counter: usize = 0;
    while (try walker.next()) |entry| {
        defer counter += 1;
        // Process only header files.
        if (!std.mem.endsWith(u8, entry.basename, ".h")) continue;

        // Compute the candidate include directory.
        const parent_dir = std.fs.path.dirname(entry.path) orelse continue;
        const raw_include_dir = b.pathJoin(&.{ comp, parent_dir });
        const trimmed_include_dir = trim_to_include(raw_include_dir);

        // Only add if the trimmed path ends with "include".
        if (!std.mem.endsWith(u8, trimmed_include_dir, "include")) continue;

        // Avoid duplicates.
        if (added.contains(trimmed_include_dir)) continue;
        try added.put(trimmed_include_dir, {});

        lib.addSystemIncludePath(.{ .cwd_relative = trimmed_include_dir });
        std.debug.print("Adding include path [{d}]: {s}\n", .{ counter, trimmed_include_dir });
    }
    std.debug.print("Added {d} include paths\n", .{counter});
}

// Helper: Given a candidate path, look for "/config/include/" first;
// then for "/include/"; return the substring up to and including the marker.
fn trim_to_include(path: []const u8) []const u8 {
    const configIncMarker = "/config/include/";
    if (std.mem.indexOf(u8, path, configIncMarker)) |i| {
        return path[0 .. i + configIncMarker.len - 1];
    }
    const incMarker = "/include/";
    if (std.mem.indexOf(u8, path, incMarker)) |i| {
        return path[0 .. i + incMarker.len - 1];
    }
    if (std.mem.endsWith(u8, path, "/include")) {
        return path;
    }
    return path;
}
```
</details>

emit:

```bash
Build Summary: 0/4 steps succeeded; 1 failed
install transitive failure
└─ install zig_main_exe transitive failure
   └─ zig build-exe zig_main_exe Debug xtensa-freestanding-none transitive failure
      └─ translate-c 1 errors
/home/kassane/esp/v5.4/esp-idf/components/freertos/FreeRTOS-Kernel/include/freertos/FreeRTOS.h:63:10: error: 'FreeRTOSConfig.h' file not found
#include "FreeRTOSConfig.h"
         ^
```
I have yet to pull up some folders ( cannot say if zig **limits** includePath support) from esp32-riscv32 to xtensa target or vice versa. However, just xtensa-cpu and risv32-esp32p4 has `@tagName(target-cpu)` unlike `generic_rv32`.

wiiggee1, 2025-02-08:
@kassane Yeah something is weird, can seem to get it to work either. I have notice that deeper directory paths fails while less deeper seems to work. For instance I can access the `$IDF_PATH/components/hal/include/hal_utils.h`, but not other directories that have deeper include paths such as the `freertos` dir.

wiiggee1, 2025-02-09:
Any further progess? @kassane

kassane, 2025-02-09:
> Any further progess? [@kassane](https://github.com/kassane)

No. My suggestion replace `searched_idf_include` to:

```bash
tree .zig-cache 
.zig-cache
├── h
│   ├── 3e6d69d71ec3f7101b2082196c54028a.txt
│   ├── 74127c67731498f788e834228ddcc038.txt
│   ├── d46d0fa9fe56a9a304d29c38d1dcba91.txt
│   └── timestamp
├── o
│   ├── 05e0e3a9f7947fcd2561de5b4284c446
│   │   └── zig_main_exe.o
│   ├── 251b371f9e023b0a46a0237c4623c0b5
│   │   └── bindings.zig
│   ├── e1a2413a2ed45197b8bfe5625786a2c2
│   │   └── dependencies.zig
│   └── fdb573a8168cae6b79975ca6530d5ac2
│       ├── build
│       └── build.o
├── tmp
└── z
    ├── 7574f80f45d43175448f53c54cc75696
    └── 96268061beae79790ed05aa636199def

9 directories, 11 files
```
<details>
<symmary>build.zig</summary>

```zig
 bindgen.addIncludePath(.{ .cwd_relative = b.fmt("{s}/components/freertos/FreeRTOS-Kernel/include", .{esp_idf_path}) });
    bindgen.addIncludePath(.{ .cwd_relative = b.fmt("{s}/components/freertos/config/include/freertos", .{esp_idf_path}) });
    bindgen.addIncludePath(.{ .cwd_relative = b.fmt("{s}/components/freertos/config/xtensa/include", .{esp_idf_path}) });
    bindgen.addIncludePath(.{ .cwd_relative = b.fmt("{s}/components/freertos/FreeRTOS-Kernel-SMP/portable/xtensa/include/freertos", .{esp_idf_path}) });
    bindgen.addIncludePath(.{ .cwd_relative = b.fmt("{s}/components/esp_hw_support/include", .{esp_idf_path}) });
    bindgen.addIncludePath(.{ .cwd_relative = b.fmt("{s}/components/soc/esp32s3/include", .{esp_idf_path}) });
    bindgen.addIncludePath(.{ .cwd_relative = b.fmt("{s}/components/esp_common/include", .{esp_idf_path}) });
    bindgen.addIncludePath(.{ .cwd_relative = b.fmt("{s}/components/xtensa/include", .{esp_idf_path}) });
    bindgen.addIncludePath(.{ .cwd_relative = b.fmt("{s}/components/xtensa/esp32s3/include", .{esp_idf_path}) });
    bindgen.addIncludePath(.{ .cwd_relative = b.fmt("{s}/components/soc/esp32s3/register", .{esp_idf_path}) });
    bindgen.addIncludePath(.{ .cwd_relative = b.fmt("{s}/components/esp_system/include", .{esp_idf_path}) });
    bindgen.addIncludePath(.{ .cwd_relative = b.fmt("{s}/components/newlib/include", .{esp_idf_path}) });
    bindgen.addIncludePath(.{ .cwd_relative = b.fmt("{s}/components/newlib/platform_include/sys", .{esp_idf_path}) });
    bindgen.addIncludePath(.{ .cwd_relative = b.fmt("{s}/components/newlib/platform_include", .{esp_idf_path}) });
    bindgen.addIncludePath(.{ .cwd_relative = b.fmt("{s}/components/hal/platform_port/include", .{esp_idf_path}) });
    bindgen.addIncludePath(.{ .cwd_relative = b.fmt("{s}/components/heap/include", .{esp_idf_path}) });
    bindgen.addIncludePath(.{ .cwd_relative = b.fmt("{s}/components/esp_rom/include", .{esp_idf_path}) });
    bindgen.addIncludePath(.{ .cwd_relative = b.fmt("{s}/components/esp_wifi/include", .{esp_idf_path}) });
    bindgen.addIncludePath(.{ .cwd_relative = b.fmt("{s}/components/esp_event/include", .{esp_idf_path}) });
    bindgen.addIncludePath(.{ .cwd_relative = b.fmt("{s}/examples/protocols/linux_stubs/esp_stubs/include", .{esp_idf_path}) });
    bindgen.addIncludePath(.{ .cwd_relative = b.fmt("{s}/components/lwip/include", .{esp_idf_path}) });
    bindgen.addIncludePath(.{ .cwd_relative = b.fmt("{s}/components/lwip/lwip/src/include", .{esp_idf_path}) });
    bindgen.addIncludePath(.{ .cwd_relative = b.fmt("{s}/components/lwip/port/include", .{esp_idf_path}) });
    bindgen.addIncludePath(.{ .cwd_relative = b.fmt("{s}/components/freertos/config/include", .{esp_idf_path}) });
    bindgen.addIncludePath(.{ .cwd_relative = b.fmt("{s}/components/lwip/port/freertos/include", .{esp_idf_path}) });
    bindgen.addIncludePath(.{ .cwd_relative = b.fmt("{s}/components/lwip/port/esp32xx/include", .{esp_idf_path}) });
    bindgen.addIncludePath(.{ .cwd_relative = b.fmt("{s}/components/log/include", .{esp_idf_path}) });
    bindgen.addIncludePath(.{ .cwd_relative = b.fmt("{s}/components/nvs_flash/include", .{esp_idf_path}) });
    bindgen.addIncludePath(.{ .cwd_relative = b.fmt("{s}/components/esp_partition/include", .{esp_idf_path}) });
    bindgen.addIncludePath(.{ .cwd_relative = b.fmt("{s}/components/esp_netif/include", .{esp_idf_path}) });
    bindgen.addIncludePath(.{ .cwd_relative = b.fmt("{s}/components/esp_event/include", .{esp_idf_path}) });
```
</details>

wiiggee1, 2025-02-11:
@kassane Hello running the above, I managed to get it to work after struggling with MACROS not being defined. However, I now get the `bindings.zig` from c-translate, to be saved in `.zig-cache`. But in my main.zig I cant find the module when I try to import it.

kassane, 2025-02-11:
> But in my main.zig I cant find the module when I try to import it.

```zig
// replace '@cImport' to const c = '@import("idf")' zig_main_exe.root_module.addImport("idf", bindings.module("idf"));
```

wiiggee1, 2025-02-12:
@kassane So I have managed to generate the file, but I dunno if its `zls` not detecting. It does compile and everything. But I can find or see the imports in my lsp. Do you have any suggestions. Could it be due to how I setup my `addModules`. Given: 

```zig 
const std = @import("std");
const buliltin = @import("builtin");

// Although this function looks imperative, note that its job is to
// declaratively construct a build graph that will be executed by an external
// runner.
pub fn build(b: *std.Build) !void {
    // Standard target options allows the person running `zig build` to choose
    // what target to build for. Here we do not override the defaults, which
    // means any target is allowed, and the default is native. Other options
    // for restricting supported target set are available.
    const esp32s3_target = std.Target.Query{
        .cpu_arch = .xtensa,
        .cpu_model = .{ .explicit = &std.Target.xtensa.cpu.esp32s3 },
        .os_tag = .freestanding,
        .abi = .none,
    };

    // const target = b.resolveTargetQuery(esp32s3_target);
    const target = b.standardTargetOptions(.{ .default_target = esp32s3_target });
    const optimize = b.standardOptimizeOption(.{});

    // Ensure that Zig can find the necessary ESP-IDF header files.
    const home_directory = std.process.getEnvVarOwned(b.allocator, "HOME") catch "";
    const esp_idf_path = std.process.getEnvVarOwned(b.allocator, "IDF_PATH") catch "";
    // const src_path = std.fs.path.dirname(@src().file) orelse b.pathResolve(&.{"./src/"});

    // generate bindings for the ESP-IDF library
    const bindings = b.addTranslateC(.{
        .link_libc = true,
        .optimize = optimize,
        .target = target,
        .root_source_file = b.path("bindings.h"),
    });

    bindings.defineCMacro("__xtensa", "");
    bindings.defineCMacro("__COUNTER__", "0");
    bindings.defineCMacro("CONFIG_IDF_TARGET_ESP32S3", "1");
    bindings.defineCMacro("XCHAL_NUM_AREGS", "64");
    bindings.defineCMacro("XCHAL_HAVE_s32c1I", "1");

    //Here goes a static library to be linked with c / c++ app.
    const esp_idf_lib = b.addStaticLibrary(.{
        .link_libc = true,
        .name = "zig_main",
        .root_source_file = b.path("src/main.zig"),
        .target = target,
        .optimize = optimize,
    });

    const include_paths = [_][]const u8{
        "components/freertos/FreeRTOS-Kernel/include",
        "components/freertos/config/include/freertos",
        "components/freertos/config/xtensa/include",
        "components/freertos/FreeRTOS-Kernel-SMP/portable/xtensa/include/freertos",
        "components/esp_hw_support/include",
        "components/soc/esp32s3/include",
        "components/esp_common/include",
        "components/xtensa/include",
        "components/xtensa/include/xtensa",
        "components/xtensa/esp32s3/include",
        "components/xtensa/esp32s3/include/xtensa/config",
        "components/soc/esp32s3/register",
        "components/esp_system/include",
        "components/newlib/include",
        "components/newlib/platform_include/sys",
        "components/newlib/platform_include",
[... 218 lines cut ...]

pub fn searched_idf_libs(b: *std.Build, lib: *std.Build.Step.Compile) !void {
    var dir = try std.fs.cwd().openDir("../build", .{
        .iterate = true,
    });
    defer dir.close();
    var walker = try dir.walk(b.allocator);
    defer walker.deinit();

    while (try walker.next()) |entry| {
        const ext = std.fs.path.extension(entry.basename);
        const lib_ext = inline for (&.{".obj"}) |e| {
            if (std.mem.eql(u8, ext, e))
                break true;
        } else false;
        if (lib_ext) {
            const src_path = std.fs.path.dirname(@src().file) orelse b.pathResolve(&.{".."});
            const cwd_path = b.pathJoin(&.{ src_path, "build", b.dupe(entry.path) });
            const lib_file: std.Build.LazyPath = .{ .cwd_relative = cwd_path };
            lib.addObjectFile(lib_file);
        }
    }
}

```

kassane, 2025-02-12:
@wiiggee1,
What version of zls? Has it been build using zig-xtensa?
zig-upstream doesn't have full xtensa and zls can read the builtin from the toolchain.

kassane, 2025-02-12:
@wiiggee1 
[image]

wiiggee1, 2025-02-12:
@kassane Ohhh might have been ZLS all along. I haven't rebuilt it using the zig Xtensa compiler. Have used the ZLS using neovim and Mason. 

I'll give this a try!

kassane, 2025-02-12:
> But did you just build ZLS with the xtensa zig?

Exactly!

```bash
git clone --recursive https://github.com/zigtools/zls
cd zls
# downgrade (zig-xtensa compat)
git checkout d3d11a089d608014d50aca429fcb8bd82724fedf
$ZIG_XTENSA_PATH/zig build -Doptimize=ReleaseFast
```

wiiggee1, 2025-02-13:
@kassane After some tinkering, and modifying my `ZLS` config with correct builds and paths, I now seem to get module definition from the lsp. I'll love to keep this thread open for discussions and potential other challenges I might face a long the way. 

Thanks for the fast answers and help Kassane. I appreciate it!

kassane, 2025-02-13:
You're welcome! This is the role of the maintainer, and thanks for your trust in my work (which is really just a volunteer experiment).

If the problem persists, just comment and I'll reopen the issue!

closed 2025-02-13 by kassane