kassane/zig-esp-idf-sample #27: Not building: No rule to make target 'lib/libapp_zig.a', needed by 'zig-sample-idf.elf'.

MainManu, 2025-02-17:
I cannot get idf.py to build the sample. 

## Info

Zig version:  0.13.0
idf.py version: ESP-IDF v5.5-dev-1898-g472792eb8a-dirty
Target: ESP32

## Steps to reproduce

1. clone the repo
2. install latest idf.py
3. set target ESP32 (idf.py set-target esp32)
4. build (idf.py build)

## Error

`
make[2]: *** No rule to make target 'lib/libapp_zig.a', needed by 'zig-sample-idf.elf'.  Stop.
make[1]: *** [CMakeFiles/Makefile2:2564: CMakeFiles/zig-sample-idf.elf.dir/all] Error 2
make: *** [Makefile:136: all] Error 2
`

kassane, 2025-02-18:
Hi @MainManu,

Missing more information... Please, show full-log.
Have you already tried idf v5.4?

kassane, 2025-02-19:
I tested idf-master and I didn't get the same problem you presented.

MainManu, 2025-02-19:
Thanks for the quick reply @kassane . As I am on the move rn, I tried again on a macbook with the exact same zig and IDF version, and for some reason it built without issues. Seems to be related to the gcc or ninja versions. Will post the full logs tomorrow.

MainManu, 2025-02-24:
make Version with error:
GNU Make 4.3

make version without error:
GNU Make 3.81

MainManu, 2025-02-24:
cmake versions identical:
3.31.5

kassane, 2025-02-24:
The log shown above is not very clear. What is the operating system and architecture?

Can you confirm if the zig tarball file was extracted into the `$PWD/build` folder?

Please! Try generating a log file and send it to me as an attachment.
```bash
idf.py build 2>&1 > stdout.log
```

kassane, 2025-02-24:
> Zig version: 0.13.0

By default zig-xtensa (via CMake downloader) is v0.14.0-dev (final release coming soon).

However, if you modify the `cmake/zig-build.cmake` script you can use the compiler installed on your OS.

closed 2025-03-09 by kassane